Lotlab/nrf52-keyboard #17: 你好！在编译USB固件时出现以下问题

ada19911201, 2019-09-09:
../../usb/usb.mk:41: recipe for target '_build/ch554' failed
make: [_build/ch554] Error 1 (ignored)
Compiling usb file: endpoints.c
../../usb/endpoints.c:60: error 4: 'fopen' failed on file '_build/ch554\endpoints.asm'
../../usb/usb.mk:48: recipe for target '_build/ch554/endpoints.rel' failed
make: *** [_build/ch554/endpoints.rel] Error 1

jiangming1399, 2019-09-09:
编译环境？

ada19911201, 2019-09-09:
我是在W10下配置的编译环境，运行SDCC -V没问题，不知道是不是WIN10的原因。也下载了CodeBlocks进行编译也不行，找不到config.h文件
USB固件也是在\keyboard\lot60-ble目录下编译对吧?

jiangming1399, 2019-09-09:
是的。Win10的什么环境下编译的？是CMD，MSYS，POWERSHELL还是WSL？CMD和POWERSHELL下我个人没试过，你看可不可以用MSYS编译试试

ada19911201, 2019-09-09:
好的，我用的是CMD,可能会有问题，我试试MSYS下可以吗，谢谢了

ada19911201, 2019-09-10:
解决了，要在编译地方建一个CH554文件夹，就不会报错了

jiangming1399, 2019-09-10:
这么奇怪的错误吗……

jiangming1399, 2019-09-23:
具体的原因是linux和cmd的mkdir的语法不一样，正斜杠在windows下不识别，导致mkdir失败。这个问题我之后有可能就修吧

closed 2019-09-23 by jiangming1399