Lotlab/nrf52-keyboard #47: nrf52832的编译问题

newtryk, 2020-08-27:
我的环境是mac系统 docker 下的 ubuntu 20.04的环境, 按照你wiki的文档操作. 在lkb-core 这个项目下 用nrf52832芯片的情况下编译是没问题的, 并且已经烧录到硬件上 能成功连接到电脑了 貌似没什么问题. 可是在其他项目下  比如lot60-ble/rev_f, Omega45, Omega50 , 如果使用 nrf52810则编译成功, 如果把芯片改成nrf52832 则编译失败, 报错如下:

> Compiling file: nrf_svc_handler.c
Compiling file: nrf_sdh.c
Compiling file: nrf_sdh_ble.c
Compiling file: nrf_sdh_soc.c
Compiling file: oberon_backend_chacha_poly_aead.c
Compiling file: oberon_backend_ecc.c
Compiling file: oberon_backend_ecdh.c
Compiling file: oberon_backend_ecdsa.c
Compiling file: oberon_backend_eddsa.c
Compiling file: oberon_backend_hash.c
Compiling file: oberon_backend_hmac.c
Linking target: /data/nrf52-keyboard/keyboard/Omega50/_build/nrf52_bootloader.out
/usr/local/gcc-arm-none-eabi-7-2018-q2-update/bin/../lib/gcc/arm-none-eabi/7.3.1/../../../../arm-none-eabi/bin/ld: error: /data/nrf52-keyboard/keyboard/Omega50/_build/nrf52_bootloader.out uses VFP register arguments, ../../../SDK/external/micro-ecc/nrf52nf_armgcc/armgcc/micro_ecc_lib_nrf52.a(uECC.c.o) does not
/usr/local/gcc-arm-none-eabi-7-2018-q2-update/bin/../lib/gcc/arm-none-eabi/7.3.1/../../../../arm-none-eabi/bin/ld: failed to merge target specific data of file ../../../SDK/external/micro-ecc/nrf52nf_armgcc/armgcc/micro_ecc_lib_nrf52.a(uECC.c.o)
collect2: error: ld returned 1 exit status
make[1]: *** [../../../template/Makefile.common:293: /data/nrf52-keyboard/keyboard/Omega50/_build/nrf52_bootloader.out] Error 1
make[1]: Leaving directory '/data/nrf52-keyboard/application/bootloader/project'
make: *** [../../application/main/project/bl.mk:19: bootloader] Error 2

好像是 nrf52_bootloader.out 这个没有生成, 不知道是什么问题. 谢谢~

jim-kirisame, 2020-08-27:
52832和52810的bootloader用的uECC库是不一样的。

closed 2020-08-27 by jim-kirisame

newtryk, 2020-08-27:
但是在 lkb-core 这个库下就能编译过去, 这个有点...

newtryk, 2020-08-27:
好吧, 那这个问题怎么解决呢? 是要找52832 库吗? 还是项目里有个配置参数可以改? 
ps:大佬回复的真速度!!!