Lotlab/nrf52-keyboard #48: nrf52832的编译问题

newtryk, 2020-08-28:
hello 又是我, 上一个问题提到的 52832 编译的问题 我多次尝试之后发现 报错是跟一个配置有关  就是这个
> NRF52_DISABLE_FPU = yes # 禁用FPU

如果使用52832的主控 但是又开着这个fpu 编译bootloader的时候就会报错, 如果是 52810 就没关系, 不知道是我的环境有问题.
另外还想问一下 禁用这个有什么影响. 上个issue 关掉了, 然后大佬后来又没回复了, 所以我就新建了一个 希望大佬不要介意!

jim-kirisame, 2020-08-28:
github的邮件回复有问题。

因为关了硬浮点

在 2020-08-27 10:45, newtryk 写道:
> 但是在 lkb-core 这个库下就能编译过去, 这个有点...

编译对应的uECC库，或者关掉硬浮点。

在 2020-08-27 10:53, newtryk 写道:
> 好吧, 那这个问题怎么解决呢? 是要找52832 库吗? 还是项目里有个配置参数可以改?
> ps:大佬回复的真速度!!!

closed 2020-08-28 by jim-kirisame

SunRain, 2020-11-06:
@jim-kirisame 
请问52832对应的uecc库在哪里，不是/SDK/external/micro-ecc/nrf52nf_armgcc/armgcc/吗（或者是nrf52hf_armgcc这个？）

> 编译对应的uECC库，或者关掉硬浮点。
> 
> 在 2020-08-27 10:53, newtryk 写道:
> 
> > 好吧, 那这个问题怎么解决呢? 是要找52832 库吗? 还是项目里有个配置参数可以改?
> > ps:大佬回复的真速度!!!