Lotlab/nrf52-keyboard #53: 关于nrf52-keyboard固件烧录的一些问题

SunRain, 2020-11-04:
你好，
    参照您在github上的这个项目，我在自己尝试制作一个蓝牙键盘，目前pcb已经制作完成，固件也编译完了，但是在烧录的时候有一些步骤不太了解，希望能得到您的帮助，具体如下：
    参照https://wiki.lotlab.org/上的相关文档，我使用的是ch554t的芯片以及nrf52832的蓝牙芯片，均购买自淘宝散件，其中nrf52832使用的是易佰特的E73-2G4M04S1B（https://detail.tmall.com/item.htm?spm=a1z10.5-b-s.w4011-21814129248.67.3aa0475faiYeUE&id=597243668972&rn=c830af448ac18f69fab5bd968f211087&abbucket=3&skuId=4157008284764）
    1.ch554的芯片烧录步是
       参照Lot60-BLE-Keyboard的电路图，短接ch554上的k1按钮，插入电脑usb口，断开k1，使用烧录工具烧录ch554固件
    2.linux下使用pyocd和dap link烧录蓝牙芯片
       a.daplink 连接蓝牙的swd口（没有连接3v3引脚，直接usb连接电脑，即使用电脑usb口供电，而非dap link的3v3口供电）
       b.使用pyocd烧录, pyocd flash -t nrf52 path/to/my/nrf-bootloader.hex
                                pyocd flash -t nrf52 path/to/my/Lot60-BLE-Keyboard-fw.hex
不知道上面蓝牙烧录步骤是否有问题，在使用swd烧录的时候，需不需要先短接nrf52的reset引脚（即，先短接reset引脚，然后连接usb）

jim-kirisame, 2020-11-04:
没有问题。如果不工作，请检查是否烧录了softdevice；如果烧录了bootloader，则再检查是否烧录了签名（setting)

closed 2020-11-04 by jim-kirisame

SunRain, 2020-11-04:
请问Setting是哪部分的内容(固件)?

jim-kirisame, 2020-11-04:
是固件签名，通过make setting 生成。

SunRain, 2020-11-05:
您好,刚刚又发现一个问题.之前我是按照wiki上面编译的,直接跳过了项目README文件里面的单独编译bootload的内容(即,编译完uEcc库后,直接cd keyboard/lot60-ble,编译里面的bootload),这次我按照项目README里面的步骤,cd application/bootloader/project以后编译bootload,提示

> ../sdk_config.h:47:10: fatal error: config.h: No such file or directory .   #include "config.h"

在sdk_config.h开头,有说明

> // <<< Use Configuration Wizard in Context Menu >>>\n
#ifdef USE_CUSTOM_CONFIG
#include "config.h"

看注释似乎是IDE里面配置以后添加自定义配置的宏,请问这个能否直接注释掉(注释掉后可编译通过)

@jim-kirisame

jim-kirisame, 2020-11-05:
此方法已过期,请不要单独编译bootloader了