Lotlab/nrf52-keyboard #59: 自编译固件不运行

ThermalEng, 2020-11-29:
大神求助！
看到develop分支的新特性后，想做个新键盘，参考[nrfmicro](https://oshwhub.com/Wayne654/nrfmicro)自己打了个板子，主控是RF-BM_ND04L，usb芯片是ch552T。
develop分支下固件成功编译，isp模式刷入usb固件，用手机dfu模式下刷入bootloader和application，（bootloader是用nrfutil把nrf52_bootloader.hex打包成zip）。键盘插入usb后，电脑可以显示键盘名称，但无法输出，ble、usb、chrg指示灯均不亮。蓝牙无法被搜到，nrfconnect可连接。
尝试用pyOCD连接，显示0   Lotlab CMSIS-DAP   44C79A55，但刷固件返回0002669:CRITICAL:__main__:No ACK received。
在config.h添加代码
`#define SWD_DAT_IO T2
#define SWD_DAT_MASK bT2
#define SWD_DAT_PORT P1
#define SWD_CLK_IO T2EX
#define SWD_CLK_MASK bT2EX
#define SWD_CLK_PORT P1`
重新编译、刷入，结果仍一样。
这是什么问题，是否是bootloader原因？应该怎么解决？
[键盘设置文件](https://github.com/ThermalEng/keyboard/)

jim-kirisame, 2020-11-29:
你能够手机DFU就说明bootloader已经正确烧录了，不需要重新烧。
蓝牙如果可以搜索到的话，说明固件是在正常工作了，如果不能连接或是其他问题，建议擦除后重新刷一次固件。

使用pyOCD的话，需要连线到SWD口，请根据配置自行连线。

ThermalEng, 2020-11-29:
嗯，插上usb后usb指示灯不亮，ble1灯间歇性闪烁。估计不是固件问题，就是和usb通信有问题。调试器在路上，收到后swd口擦除后刷一次再反馈。

genokolar, 2020-11-30:
注意刷入setting

closed 2021-04-12 by jim-kirisame