Lotlab/nrf52-keyboard #64: lot-60更改nrf52832主控编译的固件不能运行

15230745073, 2020-12-25:
使用lot60键盘的rev.g版本程序，在其基础上更改common.mk中的主控为52832后编译固件完成，然后将固件烧录到52832模块中，搜不到蓝牙信号。但是使用core的程序或者Omega系列的程序自编译固件烧录到52832模块中可以运行并搜到蓝牙信号。
使用的烧录方式为Jlink烧录蓝牙完整程序包，操作过程为先make编译主程序然后make bootloader编译bootloader然后merge_all生成seething签名并打包，然后使用Jlink烧录打包完成的nrf52_kbd_sign_with_sd.hex固件。

jim-kirisame, 2020-12-25:
烧录bootloader了吗？

15230745073, 2020-12-25:
非常抱歉，重新检查了一下单独烧录bootloader然后烧录完整包的语句有点问题，我将打包的语句调整为将bootloader也一起打包然后烧录，之后运行正常，感谢作者。

15230745073, 2021-01-11:
这段时间没有搞，今天重新烧录打包了bootloader一起的固件又没有蓝牙信号了，希望得到作者的帮助。现在的固件包是打包了softdevice-bootloader-application-setting

jim-kirisame, 2021-01-11:
操作没有问题。

15230745073, 2021-01-11:
辛苦作者出一版52832的lot60固件呗

15230745073, 2021-01-11:
我现在不太确定是我这边编译环境有问题还是咱们的程序有问题

jim-kirisame, 2021-01-11:
暂时没有出对应的硬件的计划，所以也不会出对应的固件。

jim-kirisame, 2021-01-11:
程序方面是没有问题的。如果不确定是哪方面的问题，建议先擦除芯片然后重新烧录softdevice和application。bootloader并不是必须的，你可以先试着寻找问题的来源。

15230745073, 2021-01-11:
我这边编译core和omega系列的固件烧录以后是没问题的，可是lot60就有问题

15230745073, 2021-01-16:
希望作者开源一版keil的工程😆平时总用keil，GDB有点搞不明白

jim-kirisame, 2021-01-16:
不会做这个的支持。

Keil缺少一项编译需要的重要功能，所以暂时以及之后都不会做这方面的支持。

15230745073, 2021-01-16:
好的感谢作者，感谢作者的辛苦付出

closed 2021-01-20 by jim-kirisame