Lotlab/nrf52-keyboard #67: 编译dfu包错误

jeromeli002, 2021-02-15:
安装教程一步一步做的 
到最后编译duf包的时候出错了 
是哪儿没对？？？
Compiling file: nrf_sdh_ble.c
Compiling file: nrf_sdh_soc.c
Compiling file: app_timer.c
Linking target: _build/nrf52_kbd.out
   text	   data	    bss	    dec	    hex	filename
  54068	    992	   5828	  60888	   edd8	_build/nrf52_kbd.out
Preparing: _build/nrf52_kbd.hex
Preparing: _build/nrf52_kbd.bin
DONE nrf52_kbd
Packing: _build/nrf52_kbd.hex
nrfutil pkg generate --hw-version 52 --application-version 1 --application _build/nrf52_kbd.hex \
--sd-req 0xb7 --key-file ../../application/main/project/private.key _build/nrf52_kbd_77050eb1.zip
make: nrfutil: 没有那个文件或目录
make: *** [../../application/main/project/app.mk:350：package] 错误 127

jim-kirisame, 2021-02-16:
make bootloader
make merge_all
具体请看make help

closed 2021-02-16 by jim-kirisame

jeromeli002, 2021-03-12:
Merging program, signature and SoftDevice s112_nrf52_6.1.1_softdevice.hex to _build/nrf52_kbd_sign_with_sd.hex
mergehex -m ../../SDK/components/softdevice/s112/hex/s112_nrf52_6.1.1_softdevice.hex _build/nrf52_settings.hex _build/nrf52_kbd.hex -o _build/nrf52_kbd_sign_with_sd.hex
make: mergehex: 没有那个文件或目录
make: *** [../../application/main/project/app.mk:375：merge_all] 错误 127

到最后出错了 nrf52_kbd_sign_with_sd.hex按提示发现没有这个文件

jim-kirisame, 2021-03-12:
装mergehex

jeromeli002, 2021-03-13:
编译指南是没有更新还是有错误？？
编译bootloader这步不一样？？
```
# 方法1：
   cd application/bootloader/project/armgcc/
   make SOFTDEVICE=S132 NRF_CHIP=nrf52832 NRF52_DISABLE_FPU=yes -j # nrf52832的编译命令
project/下没有 armgcc 这个目录。如果只进project/ 再执行下一步命令就一堆错误
```
如果用这样
 ```
# 方法2：
   cd keyboard/template
   make bootloader -j

只有tempplate和lkb-core 目录才编译成功其他目录的都是失败
如果把52832改成52810也就能编译通过

_Compiling file: oberon_backend_hash.c
Compiling file: oberon_backend_hmac.c
Linking target: /home/jl/nrf52-keyboard/keyboard/newhope64/_build/nrf52_bootloader.out
arm-none-eabi-gcc: error: ../../../SDK/external/micro-ecc/nrf52hf_armgcc/armgcc/micro_ecc_lib_nrf52.a: No such file or directory
make[1]: *** [../../../template/Makefile.common:293：/home/jl/nrf52-keyboard/keyboard/newhope64/_build/nrf52_bootloader.out] 错误 1
make[1]: 离开目录“/home/jl/nrf52-keyboard/application/bootloader/project”
make: *** [../../application/main/project/bl.mk:19：bootloader] 错误 2_

```

这是哪步出错了？？？

jim-kirisame, 2021-03-13:
自行修改路径编译uecc库

jeromeli002, 2021-03-13:
git submodule init -- # 初始化子模块，用于构建microecc库
   cd SDK/external/micro-ecc/nrf52nf_armgcc/armgcc
   make # 编译microecc库
   cd ../../../../../

是这个吗？
这个已经改过了没有报错 下一步就不行了

jim-kirisame, 2021-03-14:
我已经更新了教程，跟着里面走应该就没问题了。

jeromeli002, 2021-03-14:
感谢大佬指导小白 
 编译过了