Lotlab/nrf52-keyboard #75: make：~/.local/gcc-arm-none-eabi-7-2018-q2-update/bin/arm-none-eabi-gcc：命令未找到

Ozol-alt, 2021-03-26:
大神求助！
我根据[LotKB固件编译移植指南](https://wiki.lotlab.org/ble/build-guide.html)，在Ubuntu 20.04.1 LTS(focal)上部署了环境
但是在/keyboard/template文件夹下make -j的时候出现如下错误：
[image]
我已经把nrf52-keyboard，gcc-arm-none-eabi-7-2018-q2-update这两个文件夹的权限改成了777，我曾今还怀疑过是gcc默认版本不对，于是我把默认版本换成了7.3.1，但还是出现同样的错误，我无法找到原因（网上也搜不到解决方案）。
请问有什么解决办法吗？
我也找到了了makefile.common里对应的127行的程序，但是我没有相关的知识，无法解读程序，下面是源码：

```shell
$(if $(shell $(CC) --version),,$(info Cannot find: $(CC).) \
  $(info Please set values in: "$(abspath $(TOOLCHAIN_CONFIG_FILE))") \
  $(info according to the actual configuration of your system.) \
  $(error Cannot continue))
```

这些程序提示了什么问题吗？求解答。

jim-kirisame, 2021-03-30:
你好，请检查一下你的gcc是不是指定的版本

closed 2021-04-10 by jim-kirisame

Ozol-alt, 2021-04-10:
首先很抱歉现在才回复，我去了一个上不了网的地方...
我很确信我是完全按照你们的wiki来搭建环境的，但是就是无法成功
[image]
我自己已经尝试过很多次了，但是都止步于最后一步，所以我真的很困惑

jim-kirisame, 2021-04-10:
看上去路径是没问题的，所以请确认截图中报错的那个路径的文件真实存在。

如果搭建环境还是有问题的话，建议使用Docker。

Ozol-alt, 2021-04-10:
非常感谢！
问题解决了，我感觉应该是之前的失败是环境搭建不够严谨导致的
tmk_core出现了不能自动下载的情况，我先使用git clone直接克隆到tmk_core文件夹然后再使用git submodule update，终于成功编译

jim-kirisame, 2021-04-11:
看你的截图里面的错误信息应该是和子模块无关的，不过还是恭喜你成功编译了