Lotlab/nrf52-keyboard #80: USB won't connect on some device

lightingghost, 2021-06-29:
I tried to connect this keyboard to a PC, but the usb connection does not work under both bios and OS. (USB light not on, keys does not work.)

However, when I connected the keyboard to a mac, usb connects without a problem.

I would like to learn how to debug this problem.

jim-kirisame, 2021-06-29:
Are you using the latest usb firmware? If not, try to upgrade it.

And have you try other usb port without usb hub? some hub might not woking

lightingghost, 2021-06-29:
> Are you using the latest usb firmware? If not, try to upgrade it.
> 
> And have you try other usb port without usb hub? some hub might not woking

Yes I have upgraded to the latest firmware. I am connecting to the PC and mac using the same hub.

genokolar, 2022-08-29:
using the latest usb firmware