Lotlab/nrf52-keyboard #81: make bootloader 编译报错

fcl999, 2021-08-06:
求帮助
lkb-core目录
```
fcl@MiBook-Pro MINGW64 /e/nrd/nrf52-keyboard/keyboard/lkb-core (master)
make bootloader
make[1]: Entering directory 'E:/nrd/nrf52-keyboard/application/bootloader/project'
make[1]: Leaving directory 'E:/nrd/nrf52-keyboard/application/bootloader/project'
Makefile:277: *** empty variable name.  Stop.
../../application/main/project/bl.mk:19: recipe for target 'bootloader' failed
make: *** [bootloader] Error 2
```
template 目录
```
fcl@MiBook-Pro MINGW64 /e/nrd/nrf52-keyboard/keyboard/template (master)
$ make bootloader
make[1]: Entering directory 'E:/nrd/nrf52-keyboard/application/bootloader/project'
make[1]: Leaving directory 'E:/nrd/nrf52-keyboard/application/bootloader/project'
Makefile:277: *** empty variable name.  Stop.
../../application/main/project/bl.mk:19: recipe for target 'bootloader' failed
make: *** [bootloader] Error 2
```

fcl999, 2021-08-06:
```c
include $(TEMPLATE_PATH)/Makefile.common
# 下面这行是277 报错的行 似乎是说define_target没有 ，但是我看了下Makefile.common是定义了的
$(foreach target, $(TARGETS), $(call define_target, $(target))) 
```

jim-kirisame, 2021-08-07:
需要更多信息

fcl999, 2021-08-07:
在windwos环境下 下载了1.1.4.2版本的代码 然后安装了Small Device C Compiler  和 gcc-arm-none-eabi 7 2018-q2-update 、Msys2
配置 template/Makefile.windows里的路径  
然后在 keyboard/lkb-core 里 改了NRF_CHIP := nrf52810 使用 make命令成功编译了kbd 和  ch554 
在SDK/external/micro-ecc/nrf52nf_armgcc/armgcc里make 编译了 armgcc
然后cd 到 keyboard/lkb-core 下 make bootloader 就报这个错误
cd到keyboard/ template 也是这个错误 就编译不成功
命令行全都是使用的 git bash here cmd

fcl999, 2021-08-07:
> 需要更多信息

在windwos环境下下载了1.1.4.2版本的代码然后安装了Small Device C Compiler和gcc-arm-none-eabi 7 2018-q2-update 、Msys2
配置模板/Makefile.windows里的路径
然后在keyboard/lkb- core 里改了NRF_CHIP := nrf52810 使用make命令成功编译了kbd和ch554
在SDK/external/micro-ecc/nrf52nf_armgcc/armgcc里make编译了armgcc
然后cd到keyboard/lkb-core下make bootloader就报这个
cd到键盘/模板也是错误 就编译不成功
命令行 全这个都在这里使用 git bash cmd

jim-kirisame, 2021-08-07:
是只有bootloader报错吗？

fcl999, 2021-08-07:
> 是只有引导加载程序报错吗？

是的 就bootloader报错

jim-kirisame, 2021-08-07:
不是很清楚问题所在，我这边在msys2下编译完全正常。

fcl999, 2021-08-07:
> 不是很清楚问题所在，我这边在msys2下编译完全正常。

方便远程看一下么 困扰我很久了 问过好多qq群QAQ

jim-kirisame, 2021-08-07:
建议尝试使用Docker环境编译，那个应该是没问题的

fcl999, 2021-08-07:
> 建议尝试使用Docker环境编译，那个应该是没问题的

没有用过docker 是装完docker 直接在cmd 下docker run lotlab/nrf52-keyboard 嘛？

fcl999, 2021-08-07:
> 建议尝试使用Docker环境编译，那个应该是没问题的

也不行呀 
```
Compiling file: oberon_backend_ecc.c
Compiling file: oberon_backend_ecdh.c
Compiling file: oberon_backend_ecdsa.c
Compiling file: oberon_backend_eddsa.c
Compiling file: oberon_backend_hash.c
Compiling file: oberon_backend_hmac.c
Linking target: ../../../keyboard/lkb-core/_build/nrf52_bootloader.out
arm-none-eabi-gcc: error: ../../../SDK/external/micro-ecc/nrf52nf_armgcc/armgcc/micro_ecc_lib_nrf52.a: No such file or directory
make[1]: *** [../../../template/Makefile.common:293: ../../../keyboard/lkb-core/_build/nrf52_bootloader.out] Error 1
make[1]: Leaving directory '/work/application/bootloader/project'
make: *** [../../application/main/project/bl.mk:19: bootloader] Error 2
#
```

jim-kirisame, 2021-08-07:
你得先编译micro ECC啊

fcl999, 2021-08-07:
好了 我把docker里的work 烤出来编译 一点问题没有 谢谢大佬

closed 2021-08-07 by fcl999