Lotlab/nrf52-keyboard #82: 烧录后没有出现蓝牙信号

fcl999, 2021-08-08:
又来了 -、-  
使用template改了一份87键配置  主要是832 改成 810 灯都注释 留了大小锁定 和 滚动锁定 修改了配列 引脚号 电源监测引脚号 
config.h
```c
#pragma once
#include <stdint.h>

/* USB和蓝牙的显示参数 */
#define VENDOR_ID 0x04d9 /* USB VID */
#define PRODUCT_ID 0xa002 /* USB PID */
// 配置协议使用的VID和PID，自行修改。ID组合不得重复。
#define CONF_VENDOR_ID 0x0001 /* 配置协议使用的VendorID */
#define CONF_PRODUCT_ID 0x4100 /* 配置协议使用的ProductID */
#define DEVICE_VER 0x0001 /* 硬件版本 */
#define MANUFACTURER "Varmilo" /* 硬件制造商，用于USB和蓝牙显示 */
#define PRODUCT "Lotlab VA87M" /* 硬件名称，用于USB和蓝牙显示 */
#define MACADDR_SEPRATOR '_' /* 蓝牙名称后地址的分隔符。若不设置则不显示蓝牙名称后面的地址 */

/* USB HID report parameter */
#define KEYBOARD_EPSIZE 8 /* 键盘上传端点大小，请不要修改 */
#define NKRO_EPSIZE 28 /* 键盘NKRO端点大小，请不要修改 */

// 定义Bootmagic/Bootcheck按键
#define BOOTMAGIC_KEY_BOOT KC_U /* 开机 */ 
#define BOOTMAGIC_KEY_ERASE_BOND KC_E /* 删除所有绑定 */

// 键盘省电参数
#define SLEEP_SLOW_TIMEOUT 15 // 键盘闲置多久后转入慢速扫描模式 (s)
#define SLEEP_OFF_TIMEOUT 1200 // 键盘闲置多久后转入自动关机 (s)
#define KEYBOARD_SCAN_INTERVAL 1 // 键盘最小时间单位TICK (ms)
#define KEYBOARD_FAST_SCAN_INTERVAL 10 // 通常模式下，多久扫描一次键盘 (ms)
#define KEYBOARD_SLOW_SCAN_INTERVAL 100 // 慢速模式下，多久扫描一次键盘 (ms)
#define LED_AUTOOFF_TIME 5 /* LED自动熄灭时长(s)，设为0则不自动熄灭 */

// 键盘额外功能
// #define DYNAMIC_TX_POWER /* 启用自动发射功率调整 */
// #define PASSKEY_REQUIRED /* 需要输入配对码 */
#define ENABLE_WATCHDOG /* 启用看门狗 */
#define HIGH_TX_POWER /* 更改发射功率到+4dBm */
#define MULTI_DEVICE_SWITCH  /*启用多设备切换 */
#define KEYMAP_STORAGE /* 启用keymap存储 */
// #define MACRO_BLOCKING_MODE /* 在宏播放时禁用其他按键输入 */
#define MACRO_STORAGE /* 启用宏存储功能 */
#define CONFIG_STORAGE /* 启用配置存储功能 */
#define BUTTONLESS_DFU /* 启用免按钮DFU */

// #define DEBUG_SKIP_PWRON_CHECK /*  直接开机而跳过开机条件检测，用于调试 */

/* TMK固件内置功能 */
/* disable action features */
//#define NO_ACTION_LAYER
//#define NO_ACTION_TAPPING
//#define NO_ACTION_ONESHOT
//#define NO_ACTION_MACRO
//#define NO_ACTION_FUNCTION

/* Mechanical locking support. Use KC_LCAP, KC_LNUM or KC_LSCR instead in keymap */
#define LOCKING_SUPPORT_ENABLE
/* Locking resynchronize hack */
#define LOCKING_RESYNC_ENABLE

/* key combination for command */
#define IS_COMMAND() ( \
    keyboard_report->mods == (MOD_BIT(KC_LSHIFT) | MOD_BIT(KC_RSHIFT)))

// LED 配置
// #define LED_NUM **
#define LED_CAPS 30
#define LED_SCLK 31
#define LED_POSITIVE // LED上拉驱动
// #define LED_NO_DEINIT // 不要deinit端口，可以避免部分IO灯光无法关闭的问题

// Bootloader 指示灯
#define LED_DFU_INIT 30
#define LED_DFU_START 31
// #define LED_DFU_FINISH **
#define LED_DFU_POSITIVE // LED上拉驱动

// 多用途 Bootloader 按钮
// #define NRF_BL_DFU_MULTI_ROLE_BTN ** 

// USB UART 传输配置
#define HAS_USB // 启用与CH554的通信支持
#define UART_RXD 17 // UART_RX口IO
#define UART_TXD 18 // UART_TX口IO
// #define UART_DET ** // UART 检测引脚，若此脚被拉低，则说明USB正在工作。若不配置则使用RX口作为检测引脚
#define UART_BAUDRATE NRF_UART_BAUDRATE_115200 // 通信波特率，请不要修改

// 电量检测配置
#define BATTERY_ADC_PIN 2 // 电量检测引脚

// 充电检测配置
#define PIN_CHARGING !UCC1 // CH554的充电检测。当UCC1拉低时表示正在充电。若不配置则禁用USB下发充电事件
#define PIN_STANDBY !UCC2 // CH554的充电检测。当UCC2拉低时表示充电完成。若不配置则只使用PIN_CHARGING作为是否充电的检测标志

// 按键阵列配置
#define MATRIX_ROWS 6 /* 硬件阵列行数 */
#define MATRIX_COLS 17 /* 硬件阵列列数 */
static const uint8_t row_pin_array[MATRIX_ROWS] = { 8, 7, 6, 5, 4, 3 };
static const uint8_t column_pin_array[MATRIX_COLS] = { 27, 26, 25, 9, 10, 11, 12, 13, 14, 15, 16, 19, 20, 21, 22, 23, 24 };
//#define ROW_IN // 键盘阵列的二极管方向是从COL->ROW

// 剩下 28，29 gpio

/* define if matrix has ghost */
// #define MATRIX_HAS_GHOST /* 按键阵列是否出现Ghost Key，若没有加二极管则需要启用这个项目 */

#define DEBOUNCE 5 /* 硬件消抖次数，设置为0则不消抖 */
#define MATRIX_SCAN_DELAY_CYCLE 36 /* 按键扫描等待IO稳定的延时时长 */

// 板载调试器设置
#define SWD_CLK_IO T2EX
#define SWD_CLK_MASK bT2EX
#define SWD_CLK_PORT P1
#define SWD_DAT_IO T2
#define SWD_DAT_MASK bT2
#define SWD_DAT_PORT P1

```
Makefile
```makefile
# 此工程的根目录
ROOT_DIR := ../..

# 特殊目录控制
# SDK_ROOT := /var/nrf52/nRF52_SDK
# TEMPLATE_PATH := /var/nrf52/template

# 配置项：主控芯片类型
# 可选：nrf52810 或 nrf52832
NRF_CHIP := nrf52810

# 配置文件文件名
CONFIG_H = config.h
CONFIG_H_DIR = .
SRC_FILES += keymap_plain.c

# 配置项：功能选项
# 关于更多的可配置项目，请参考doc目录下的相应文档
BOOTMAGIC_ENABLE = yes	# 启用Bootmagic
BOOTCHECK_ENABLE = yes  # 启用Bootchek
MOUSEKEY_ENABLE = yes     # 启用鼠标键功能
EXTRAKEY_ENABLE = yes	# 启用媒体键功能
NKRO_ENABLE = yes	# 启用USB的全键无冲功能
USB_6KRO_ENABLE = yes # 启用USB的六键无冲功能

ONBOARD_CMSIS_DAP = yes # 启用板载调试器

########## STOP HERE ##########
# 除非你知道下面这些配置项是干什么的，否则不要去动它

INC_FOLDERS += .

APP_MAIN_DIR := $(ROOT_DIR)/application/main
TMK_CORE_DIR := $(ROOT_DIR)/tmk/tmk_core
USB_SOURCE_DIR := $(ROOT_DIR)/usb

APP_PROJ_DIR := $(APP_MAIN_DIR)/project
APP_SRC_DIR := $(APP_MAIN_DIR)/src

all: default ch554

include $(APP_PROJ_DIR)/kbd.mk
include $(USB_SOURCE_DIR)/usb.mk
```
使用`make`；`make bootloader`； `make merge_all`命令 合并出nrf52_kbd_sign_with_sd.hex 
使用pyocd + cmsisdap 烧录命令`pyocd flash --erase chip --target nrf52 nrf52_kbd_sign_with_sd.hex` 烧录成功 重启之后 搜索不到蓝牙信号，
重新编译了nrf52_kbd.hex 和 nrf52_bootloader.hex 
使用 `pyocd flash --erase chip --target nrf52 nrf52_bootloader.hex`  和  `pyocd flash --target nrf52 nrf52_kbd.hex` 还是没有信号
然后使用了keyboard中其他的示例 仅修改52832 为 52810 ，仅编译烧录nrf52_kbd.hex 使用 `pyocd flash --erase chip --target nrf52 nrf52_kbd.hex` 烧录成功 重启依旧没有蓝牙信号

jim-kirisame, 2021-08-08:
刷签名了没

fcl999, 2021-08-08:
> 刷签名了没

是密钥嘛 新键了一个public_key.c 替换了application/bootloader/dfu_public_key.c
private.key 在`make bootloader` 前 `make` 后 放到了 _build 文件夹下

jim-kirisame, 2021-08-08:
别改，直接用原厂的签名试试

fcl999, 2021-08-08:
> 别改，直接用原厂的签名试试

好的 我试试 nrf52_kbd是包含协议栈的嘛？ 我烧它的时候是全清空的状态

jim-kirisame, 2021-08-08:
不包，要自己刷

fcl999, 2021-08-08:
> 不包，要自己刷
sdk 15.3
使用 `pyocd flash --erase chip --target nrf52 s112_nrf52_6.1.1_softdevice.hex`清空并刷了协议栈 
使用官方示例 ble_app_hids_keyboard /pca10040e/s112/armgcc/ 下 make 构建 使用 `pyocd flash --target nrf52 nrf52810_xxaa.hex.hex` 刷入固件 重启 有蓝牙信号 
再次使用 `pyocd flash --erase chip --target nrf52 s112_nrf52_6.1.1_softdevice.hex`清空并刷了协议栈 
用回官方默认的签名
使用keyboard下的项目make 了新的 kbd bootloader merge_all合并  
`pyocd flash --target nrf52 nrf52_kbd_sign_with_sd.hex` 刷入
重启 无蓝牙信号

fcl999, 2021-08-08:
这个提示 里 不是说 合并了协议栈的嘛？
```
$ make merge_all
DONE nrf52_kbd
Setting generate _build/nrf52_settings.hex
nrfutil settings generate --family NRF52810 --application _build/nrf52_kbd.hex --application-version 1 --bootloader-version 1 --bl-settings-version 1 _build/nrf52_settings.hex

Note: Generating a DFU settings page with backup page included.
This is only required for bootloaders from nRF5 SDK 15.1 and newer.
If you want to skip backup page generation, use --no-backup option.

Generated Bootloader DFU settings .hex file and stored it in: _build/nrf52_settings.hex

Bootloader DFU Settings:
* File:                     _build/nrf52_settings.hex
* Family:                   NRF52810
* Start Address:            0x0002F000
* CRC:                      0xCC3C3703
* Settings Version:         0x00000001 (1)
* App Version:              0x00000001 (1)
* Bootloader Version:       0x00000001 (1)
* Bank Layout:              0x00000000
* Current Bank:             0x00000000
* Application Size:         0x0000DD10 (56592 bytes)
* Application CRC:          0x9550678F
* Bank0 Bank Code:          0x00000001
* Softdevice Size:          0x00000000 (0 bytes)
* Boot Validation CRC:      0x00000000
* SD Boot Validation Type:  0x00000000 (0)
* App Boot Validation Type: 0x00000000 (0)

Merging program, signature and SoftDevice s112_nrf52_6.1.1_softdevice.hex to _build/nrf52_kbd_sign_with_sd.hex
mergehex -m ../../SDK/components/softdevice/s112/hex/s112_nrf52_6.1.1_softdevice.hex _build/nrf52_settings.hex _build/nrf52_kbd.hex -o _build/nrf52_kbd_sign_with_sd.hex
Parsing input hex files.
Merging files.
Storing merged file.

```

jim-kirisame, 2021-08-08:
nrf52_kbd不包，带with_sd字样的才包

fcl999, 2021-08-09:
还是不行呢 nrf52_kbd_sign_with_sd 和 nrf52_settings要分开刷嘛？

jim-kirisame, 2021-08-09:
不需要。请尝试刷bootloader后再刷入nrf52_kbd_sign_with_sd

fcl999, 2021-08-09:
尝试了 还是没有信号 QAQ“=

fcl999, 2021-08-10:
> 不需要。请尝试刷bootloader后再刷入nrf52_kbd_sign_with_sd
调试了下 发现有个NRF_FAULT_ID_SDK_ERROR 是16358 然后就结束了

fcl999, 2021-08-10:
Program
 received signal SIGTRAP, Trace/breakpoint trap.
0x0001c752 in app_error_fault_handler (id=id@entry=16385, pc=pc@entry=0, info=info@entry=536895460) at ../common/../../../SDK/components/libraries/util/app_error_weak.c:100
100	    NRF_BREAKPOINT_COND;

jeromeli002, 2021-08-12:
有支持更多蓝牙的可能吗？？比如52840、52811…………

KiraLJK, 2021-09-29:
不知道你还需不需要，我也不太懂但是我自己尝试了两个办法可以用。
第一个是下载SDK以后用ses刷入，位置在SDK文件夹中的\examples\dfu\secure_bootloader\pca10040_ble\ses（这个是52832的），然后你之后就可以在手机中通过nrf connect刷入你生成的zip固件包
第二种是朋友跟我说的，在格老师那边的gt-ble60中随便找一个版本来修改配置，Jim大神这边应该没有改过，然后在\gt-ble60中make，不要在e f一类的文件夹中make，然后会生成一个600多k左右的hex文件，直接jlink啥的刷入就好。

genokolar, 2021-11-08:
大概率52810空间不足。
解决办法二个：
1、不使用bootloader，使用make merge_softdevice 生成不带bootloader的包，然后刷入nrf52_kbd_with_sd
2、使用bootloader，但修改rules.mk，去掉一些模块，比如RGB 和COMMAND。

可以关闭了

lxj5891, 2022-02-07:
手上有  jlink  pyocd  能用jlink刷吗跟debug嘛

genokolar, 2022-02-07:
jlink可以用，但是不是用pyocd。具体百度。