Lotlab/nrf52-keyboard #89: M1 Macbook Pro13 唤醒问题

NeilyoSH, 2021-09-24:
1、蓝牙配对的情况下:
USB或者蓝牙模式连接, M1 Mac休眠后唤醒, 50-70%概率会出现  不停的输入 唤醒按钮的字符, 直致 插拔USB解决.

2、删除蓝牙配对后:
type-c线连接, M1 Mac休眠后唤醒, 此时 ESC底部 RGB亮红灯, 唤醒动作无效, 必须插拔USB后, RGB灯变蓝, 此时才能唤醒Mac.

OS:    macOS 11.4
电脑: M1 Macbook Pro 13

jim-kirisame, 2021-09-24:
1. 手上没有设备暂时无法复现。群友使用同样的M1 Macbook无法复现。
2. 请确认第二种情况是ESC的RGB灯亮红灯，而不是其他设备
3. 请提供键盘硬件型号以供参考

NeilyoSH, 2021-09-26:
设备: HTKB HHKB-BLE

(请确认第二种情况是ESC的RGB灯亮红灯，而不是其他设备) 
reply: 是ESC RGB灯亮.

之前USB & 蓝牙模式  使用的时候  还出现过很多次   连击的情况(正常使用情况下)

jim-kirisame, 2021-09-26:
蓝牙模式下连击是因为信号问题，而USB下的连击我这边没有遇到过？请问发生的概率大吗？

NeilyoSH, 2021-09-28:
一直使用键盘的情况下, 每天出现 1-2次

蓝牙模式出现连击   也是    不能忍的……

NeilyoSH, 2021-09-28:
> 蓝牙模式下连击是因为信号问题，而USB下的连击我这边没有遇到过？请问发生的概率大吗？

一直使用键盘的情况下, 每天出现 1-2次  ???

NeilyoSH, 2021-11-12:
此问题在 intel iMac 27上没有复现, 

fn+left shift 切换有线模式  更无感了

closed 2021-11-12 by NeilyoSH