Lotlab/nrf52-keyboard #90: 尝试做了一个键盘，但是OLED无法使用

KiraLJK, 2021-09-26:
想问一下，按照教程制作了一个键盘，但是按照make help里面，进行了make flash_softdeivce, make flash, make flash_setting这三步后无法搜索到蓝牙信号，然后我通过SDK中的example，刷入了一个secure_bootloader的例程，可以搜索到Dfutarg，之后DFU刷入固件包后键盘可以使用，旋钮也可以使用，只有OLED无法显示。
确认了一下接线还有在config和rules.mk中的设置应该是没有问题的，想问一下是因为刷入SDK例程中的BootLoader所以有影响么？如果我希望直接使用您固件中生成的文件，我上述的刷入方式有问题么？

jim-kirisame, 2021-09-26:
前面是没问题的，由于固件存储设计Layout是需要bootloader，所以必须刷入bootloader。

OLED无法显示请确认一下接线，或是尺寸和芯片型号是否与文档中相同。

KiraLJK, 2021-09-26:
那我刷入的步骤是：make flash_softdeivce, make flash, make flash_setting，最后再用nrfjprog -f nrf52 --program _build/nrf52_bootloader.hex --sectorerase刷入了bootloader，这样是正确的么，我这样操作以后无法搜索到蓝牙的信号。我只有通过ses刷SKD里面那个example才能搜索到DFUtarg，就比较神奇

KiraLJK, 2021-09-26:
还想问一下我看了我的OLED就是128*32并且芯片是SSD1306的，无法点亮的问题可能会和skd_config这些文件中的设置有关系么？

jim-kirisame, 2021-09-26:
sdk_config内的配置请勿改动。一般来说直接开了宏接上去就好了才对的。

KiraLJK, 2021-09-26:
哦哦，那可能是因为我刷入的bootloader例子不同导致的，我对比了您程序里和我刷入的例子中的sdk_config，有些语句是不同的。

jim-kirisame, 2021-09-26:
bootloader和应用程序本体无关

KiraLJK, 2021-09-26:
所以说无论我怎么弄，只要我最后手机可以搜索到Dfutarg，并且成功的刷入了固件包，那我软件的部分就是没有问题的对么？

jim-kirisame, 2021-09-26:
是的

KiraLJK, 2021-09-26:
好的，谢谢大佬

KiraLJK, 2021-09-27:
我找到原因了，焊接连锡了。。刚好连的这俩引脚，还很小一点，不仔细看看不出，还是感谢Jim大佬

closed 2021-09-27 by KiraLJK