Lotlab/nrf52-keyboard #105: 编译好后用pyocd烧写提示intelhex.AddressOverlapError

wytalfred, 2022-05-21:
完整报错如下，有可能是什么原因呢？
D:\keyboard\pyocd> .\pyocd flash -t nrf52 ..\nrf52_kbd_with_sd.hex
0000139:WARNING:common:STLink and CMSIS-DAPv2 probes are not supported because no libusb library was found.
0000401:CRITICAL:__main__:uncaught exception: Hex file has data overlap at address 0x1000 on line 179
Traceback (most recent call last):
  File "site-packages\pyocd\__main__.py", line 368, in run
  File "site-packages\pyocd\__main__.py", line 529, in do_flash
  File "site-packages\pyocd\flash\file_programmer.py", line 157, in program
  File "site-packages\pyocd\flash\file_programmer.py", line 180, in _program_hex
  File "site-packages\intelhex\__init__.py", line 90, in __init__
  File "site-packages\intelhex\__init__.py", line 212, in loadhex
  File "site-packages\intelhex\__init__.py", line 145, in _decode_record
intelhex.AddressOverlapError: Hex file has data overlap at address 0x1000 on line 179

jim-kirisame, 2022-05-22:
检查是否芯片型号不匹配你实际刷写的型号

wytalfred, 2022-05-22:
> 检查是否芯片型号不匹配你实际刷写的型号

应该是没问题的，蓝牙芯片是 nrf52832，我是在lot60-ble/rev_g_832基础上改的

jim-kirisame, 2022-05-22:
那试着把 CUSTOM_BOOTLOADER_ADDR 这一行注释掉

wytalfred, 2022-05-22:
> 

可以了！非常感谢！！！

wytalfred, 2022-05-22:
解决

closed 2022-05-22 by wytalfred