Lotlab/nrf52-keyboard #113: 蓝牙设备从WINDOWS中删除后，再次匹配蓝牙，无法发现该设备

jackie-weiwei, 2022-07-05:
您好，我是从keyboard/template复制了一份，生成了一个IKBC的目录，config.h 如下：
`/*
Copyright 2012 Jun Wako <wakojun@gmail.com>
Copyright 2019 Jim Jiang <jim@lotlab.org>

This program is free software: you can redistribute it and/or modify
it under the terms of the GNU General Public License as published by
the Free Software Foundation, either version 2 of the License, or
(at your option) any later version.

This program is distributed in the hope that it will be useful,
but WITHOUT ANY WARRANTY; without even the implied warranty of
MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.  See the
GNU General Public License for more details.

You should have received a copy of the GNU General Public License
along with this program.  If not, see <http://www.gnu.org/licenses/>.
*/

#pragma once

#include <stdint.h>

/* USB和蓝牙的显示参数 */
// USB的VID和PID，自行修改
#define VENDOR_ID 0xEFCD /* USB VID */
#define PRODUCT_ID 0xCDEF /* USB PID */
// 配置协议使用的VID和PID，自行修改。ID组合不得重复。
#define CONF_VENDOR_ID 0xABCD /* 配置协议使用的VendorID */
#define CONF_PRODUCT_ID 0xDCBA0 /* 配置协议使用的ProductID */
#define DEVICE_VER 0x0000 /* 硬件版本 */
#define MANUFACTURER "Jackieron" /* 硬件制造商，用于USB和蓝牙显示 */
#define PRODUCT "IKBC Poker II" /* 硬件名称，用于USB和蓝牙显示 */
//#define MACADDR_SEPRATOR '_' /* 蓝牙名称后地址的分隔符。若不设置则不显示蓝牙名称后面的地址 */

/* USB HID report parameter */
#define KEYBOARD_EPSIZE 8 /* 键盘上传端点大小，请不要修改 */
#define NKRO_EPSIZE 28 /* 键盘NKRO端点大小，请不要修改 */

// 定义Bootmagic/Bootcheck按键
//#define BOOTMAGIC_KEY_BOOT KC_U /* 开机 */
//#define BOOTMAGIC_KEY_ERASE_BOND KC_E /* 删除所有绑定 */

// 键盘省电参数
#define SLEEP_SLOW_TIMEOUT 15 // 键盘闲置多久后转入慢速扫描模式 (s)
#define SLEEP_OFF_TIMEOUT 600 // 键盘闲置多久后转入自动关机 (s)
#define KEYBOARD_SCAN_INTERVAL 1 // 键盘最小时间单位TICK (ms)
#define KEYBOARD_FAST_SCAN_INTERVAL 10 // 通常模式下，多久扫描一次键盘 (ms)
#define KEYBOARD_SLOW_SCAN_INTERVAL 100 // 慢速模式下，多久扫描一次键盘 (ms)
#define LED_AUTOOFF_TIME 5 /* LED自动熄灭时长(s)，设为0则不自动熄灭 */

// 键盘额外功能
#define DYNAMIC_TX_POWER /* 启用自动发射功率调整 */
//#define PASSKEY_REQUIRED /* 需要输入配对码 */
#define ENABLE_WATCHDOG /* 启用看门狗 */
//#define HIGH_TX_POWER /* 更改发射功率到+4dBm */
#define MULTI_DEVICE_SWITCH  /*启用多设备切换 */
#define KEYMAP_STORAGE /* 启用keymap存储 */
// #define MACRO_BLOCKING_MODE /* 在宏播放时禁用其他按键输入 */
#define MACRO_STORAGE /* 启用宏存储功能 */
#define CONFIG_STORAGE /* 启用配置存储功能 */
#define BUTTONLESS_DFU /* 启用免按钮DFU */

// #define DEBUG_SKIP_PWRON_CHECK /*  直接开机而跳过开机条件检测，用于调试 */

/* TMK固件内置功能 */
/* disable action features */
//#define NO_ACTION_LAYER
//#define NO_ACTION_TAPPING
//#define NO_ACTION_ONESHOT
//#define NO_ACTION_MACRO
//#define NO_ACTION_FUNCTION

/* Mechanical locking support. Use KC_LCAP, KC_LNUM or KC_LSCR instead in keymap */
#define LOCKING_SUPPORT_ENABLE
/* Locking resynchronize hack */
#define LOCKING_RESYNC_ENABLE

/* key combination for command */
#define IS_COMMAND() ( \
    keyboard_report->mods == (MOD_BIT(KC_LSHIFT) | MOD_BIT(KC_RSHIFT)))

// LED 配置
#define LED_NUM 22
#define LED_CAPS 21
#define LED_SCLK 23
#define LED_POSITIVE // LED上拉驱动
// #define LED_NO_DEINIT // 不要deinit端口，可以避免部分IO灯光无法关闭的问题

// Bootloader 指示灯
#define LED_DFU_INIT 22
#define LED_DFU_START 23
#define LED_DFU_FINISH 24
#define LED_DFU_POSITIVE // LED上拉驱动

// 多用途 Bootloader 按钮
#define NRF_BL_DFU_MULTI_ROLE_BTN 21 

// USB UART 传输配置
//#define HAS_USB // 启用与CH554的通信支持
#define UART_RXD 17 // UART_RX口IO
#define UART_TXD 18 // UART_TX口IO
#define UART_DET 19 // UART 检测引脚，若此脚被拉低，则说明USB正在工作。若不配置则使用RX口作为检测引脚
#define UART_BAUDRATE NRF_UART_BAUDRATE_115200 // 通信波特率，请不要修改

// 电量检测配置
#define BATTERY_ADC_PIN NRF_SAADC_INPUT_AIN0 // 电量检测引脚

// 充电检测配置
#define PIN_CHARGING !UCC1 // CH554的充电检测。当UCC1拉低时表示正在充电。若不配置则禁用USB下发充电事件
#define PIN_STANDBY !UCC2 // CH554的充电检测。当UCC2拉低时表示充电完成。若不配置则只使用PIN_CHARGING作为是否充电的检测标志

// 按键阵列配置
#define MATRIX_ROWS 9 /* 硬件阵列行数 */
#define MATRIX_COLS 8 /* 硬件阵列列数 */
static const uint8_t row_pin_array[MATRIX_ROWS] = {20, 19, 18, 17, 16, 15, 14, 13, 11};
static const uint8_t column_pin_array[MATRIX_COLS] = {3, 4, 5, 6, 7, 8, 9, 10};
#define ROW_IN // 键盘阵列的二极管方向是从COL->ROW

/* define if matrix has ghost */
// #define MATRIX_HAS_GHOST /* 按键阵列是否出现Ghost Key，若没有加二极管则需要启用这个项目 */

#define DEBOUNCE 5 /* 硬件消抖次数，设置为0则不消抖 */
#define MATRIX_SCAN_DELAY_CYCLE 36 /* 按键扫描等待IO稳定的延时时长 */

// 板载调试器设置
#define SWD_CLK_IO T2EX
#define SWD_CLK_MASK bT2EX
#define SWD_CLK_PORT P1
#define SWD_DAT_IO T2
#define SWD_DAT_MASK bT2
#define SWD_DAT_PORT P1
`

编译、烧写成功后，可以匹配到该设备。当从蓝牙设备中删该设备，再次匹配则无法发现该设备，请问一下如何解决这个问题，谢谢！

jim-kirisame, 2022-07-06:
1. 关于删除后无法重新配对的问题：请清空键盘内部绑定信息，方法是开机时按住Space+E;
2. 11和13脚请自行检查硬件是否占用，并且确认芯片型号是否正确，NFC功能是否禁用等。

jackie-weiwei, 2022-07-06:
> 1. 关于删除后无法重新配对的问题：请清空键盘内部绑定信息，方法是开机时按住Space+E;
> 2. 11和13脚请自行检查硬件是否占用，并且确认芯片型号是否正确，NFC功能是否禁用等。

您好，请问一下，禁用NFC后，是不是要编译协议栈？ 我是通过 在config.h文件中 加入#define CONFIG_NFCT_PINS_AS_GPIOS来禁用的，测试好象不生效。
我开启 BOOTMAGIC_ENABLE = yes	# 启用Bootmagic 以后，搜不到蓝牙信号。

jim-kirisame, 2022-07-07:
需要，请擦除全片后重新烧录。

jackie-weiwei, 2022-07-07:
> 需要，请擦除全片后重新烧录。
> 
> 我们建议使用新的Bootcheck代替老的Bootmagic。

协议栈不知道怎么编译 好尴尬 @jim-kirisame Bootcheck代替Bootmagic 是指（BOOTMAGIC_ENABLE = no	# 启用Bootmagic
BOOTCHECK_ENABLE = yes  # 启用Bootchek ）这样设置吗？

jim-kirisame, 2022-07-07:
make flash_softdevice

jackie-weiwei, 2022-07-07:
> make flash_softdevice

@jim-kirisame 搞定了，可以用了。不过还存在一个小问题：从WINDOWS中删除设备后，不能直接清空键盘内部绑定信息，必须断电再通电，然后SPACE  + U + E 一起按住，才能清空绑定信息。清空绑定信息后，再次匹配设备，连接功后，设备在15秒左右会自动断开，此时再断电，通电后，即可正常使用了。小瑕疵，凑合用吧。
非常感谢大佬的指点，感谢，感谢！！！

jackie-weiwei, 2022-07-07:
如果大佬愿意的话，我可以提供IKBC POKER II 的配置，添加到源码，以后供大家使用。

closed 2022-07-08 by jim-kirisame