Lotlab/nrf52-keyboard #118: MacOS 无法使用openocd 进行固件更新

hi-jian, 2023-02-28:
在macos上尝试使用openocd 进行更新的时候找不到设备
```
openocd -c "adapter driver cmsis-dap" -c "init;cmsis-dap info"
```

```
Open On-Chip Debugger 0.11.0
Licensed under GNU GPL v2
For bug reports, read
	http://openocd.org/doc/doxygen/bugs.html
Error: unable to find a matching CMSIS-DAP device
```

请问有什么解决方案吗？

jim-kirisame, 2023-02-28:
有可能和这个问题类似 https://github.com/pyocd/pyOCD/issues/815，你可能需要改改pyocd的源码

hi-jian, 2023-02-28:
> 

好的，我看看。

hi-jian, 2023-03-01:
> 有可能和这个问题类似 [https://github.com/pyocd/pyOCD/issues/815，你可能需要改改pyocd的源码](https://github.com/pyocd/pyOCD/issues/815%EF%BC%8C%E4%BD%A0%E5%8F%AF%E8%83%BD%E9%9C%80%E8%A6%81%E6%94%B9%E6%94%B9pyocd%E7%9A%84%E6%BA%90%E7%A0%81)

尝试更改的过程中发现在 hid 设备列表的 path 中找不到 CMSIS-DAP 的名称，然后尝试使用配置信息进行强制链接过程发现
hid.device 方法不存在，更改成 Device后 发现open指令存在问题....

myst729, 2023-05-10:
这问题有进展吗？我某次升级 pyocd 以后也发现板载调试器不见了，后来都用的 jlink，或者 ota 升级

codesnoodles, 2024-02-27:
这个问题还有进展吗？newhope64b的板子pyocd识别不到。